STMicroelectronics/stm32l4xx-hal-driver #2: Makefile to Compile the files

heshananupama, 2020-06-01:
Do you have a Makefile to compiled all the files as a single archive/ static library? Thanks

ALABSTM, 2020-06-08:
Hi @heshananupama,

You request has been forwarded to our development teams. I will be back to you as soon as they provide me with their feedback.

With regards,

ALABSTM, 2020-07-01:
Hi @heshananupama,

I hope you are fine. Regarding your request of a Makefile, the Cube IDE already allows creating one for your project. Please follow the instructions detailed in [this](https://hackaday.com/wp-content/uploads/2017/07/tutorial.pdf) tutorial.

I hope this answers your question. Please do not hesitate if you still have something to ask.

With regards,

ALABSTM, 2020-07-20:
Hi @heshananupama,

I hope you found the tutorial I referred to in my previous comment helpful. Please allow me to close this issue now.

With regards,

closed 2020-07-20 by ALABSTM