STMicroelectronics/stm32l4xx-hal-driver #6: HAL_UARTEx_ReceiveToIdle_DMA has inconsistent behavior

gittymonkey, 2022-07-15:
When using ReceiveToIdle, there seems to be a different behavior for when IDLE and FT happen at the same time. With the HT, we get both interrupts(HT and IDLE), but when FT and IDLE happen to be at sizeof(buf), the code decides the buffer is full and cancels the idle interrupt. 

This seems strange when using the DMA in circular buffer mode, there should be no concept of a full buffer. We should get both interrupts. 

In the case when the incoming data wraps the buffer, we get the FT and IDLE.  So now there is no way to just detect IDLE interrupts. If it was consistent and we always received the IDLE interrupt, that would make it easier to discard the HT and FT.

Maybe it could check if the DMA is using CIRCULAR mode and send both  interrupts.

To explain our use case, we are receiving packets of variable size. Therefore, the size of the data cannot be used to determine when all the data is received. I am aware there are lots of ways of framing the data to detect the packets, but just being consistent with the IDLE interrupts would make it much easier.  

I have read this issue https://github.com/STMicroelectronics/stm32f1xx_hal_driver/issues/4 and read the example code, but it does not solve my issue. 

The issue also states:

>  Use of HAL_UARTEx_ReceiveToIdle_DMA service, will generate calls to
>  user defined HAL_UARTEx_RxEventCallback callback for each occurrence of
>  following events :
>  - DMA RX Half Transfer event (HT)
>  - DMA RX Transfer Complete event (TC)
>  - IDLE event on UART Rx line (indicating a pause is UART reception flow)

According to this text, we should get an interrupt for EACH occurrence, but it appears we don't always get the IDLE events.

NeoZng, 2022-10-14:
jesus i got the same problem and it cost totally two days for me to figure it out. i debugged step by step and found that it seems that there is no solution to tell which event has happened under this set of design... i have to modify the source code but once i click "generate code" in CubeMX they just gone ! so now i am using a make-shift : read the register of IDLE FLAG, which is not HAL-styled at all! it's far less elegant ...

ASELSTM, 2022-10-31:
Hi @gittymonkey, @NeoZng,
 
Thank you for this report. We have implemented a new function `HAL_UARTEx_GetRxEventType` in the frame of this [commit](https://github.com/STMicroelectronics/stm32l4xx_hal_driver/commit/9ab1cb88796fcaed1e860205622fabd737802b0a) that provides the Rx Event type (_Half transfer event_, _Transfer complete event_ or _IDL event_) which has lead to RxEvent callback execution.

As when `HAL_UARTEx_ReceiveToIdle_IT()` or `HAL_UARTEx_ReceiveToIdle_DMA()` API are called, the progress of reception process is provided to application through the calls of Rx Event callback and given that several types of events could occur, this function will allow thus to get the source of the Rx Event type that drove the call of the Rx Event callback.

With regards,

ASELSTM, 2023-04-12:
Hi @gittymonkey,

Please allow me to close this thread as no activity. You may reopen it at anytime if you have any details to share with us in order to help you to solve the issue. Thank you for your comprehension.

With regards,

closed 2023-04-12 by ASELSTM