STMicroelectronics/stm32l4xx-hal-driver #8: HAL_SPI_TransmitReceive() always returns HAL_BUSY after the first transaction

by-gnome, 2023-03-15:
In HAL_SPI_TransmitReceive() the hspi->State variable remains in HAL_SPI_STATE_BUSY_RX after the first transaction, no further transactions are executed, HAL_BUSY is returned.

It can be solved like this:
```
1545-
1546-    errorcode = HAL_ERROR;

1553-    errorcode = HAL_ERROR;

1557+    if (hspi->ErrorCode != HAL_SPI_ERROR_NONE)
1558+    {
1559+      errorcode = HAL_ERROR;
1560+    }
1561+      else
1562+    {
1563+      hspi->State = HAL_SPI_STATE_READY;
1564+    }
1565+
```

TOUNSTM, 2023-03-23:
Hello @by-gnome,

Thank you for reporting this point. Actually, this point has been discussed and fixed internally, before we receive your request. The fix will be available soon hopefully.

With Regards,
Tasnim

TOUNSTM, 2023-03-23:
ST Internal Reference: 135724

closed 2023-03-24 by TOUNSTM