STMicroelectronics/stm32l4xx-hal-driver #9: HAL_I2C_Mem_Write_IT does not correctly initialize hi2c->XferSize

thealastair, 2023-04-28:
**Describe the bug**
If an I2C write transaction fails, the hi2c->XferCount may be left set to a non zero value. Subsequent calls to HAL_I2C_Mem_Write_IT  will then ALWAYS fail because the condition hi2c->XferSize == 0U is not met in the interrupt handler

I2C_Master_ISR_IT
```
  else if ((I2C_CHECK_FLAG(tmpITFlags, I2C_FLAG_TCR) != RESET) && \
           (I2C_CHECK_IT_SOURCE(ITSources, I2C_IT_TCI) != RESET))
  {
    if ((hi2c->XferCount != 0U) && (hi2c->XferSize == 0U))
```
this causes HAL_I2C_ERROR_SIZE to be returned.
```
     {
        /* Wrong size Status regarding TCR flag event */
        /* Call the corresponding callback to inform upper layer of End of Transfer */
        I2C_ITError(hi2c, HAL_I2C_ERROR_SIZE);
      
}
```
A clear and concise description of what the bug is.

**How To Reproduce**
Perform a failing I2C transaction, then call HAL_I2C_Mem_Write_IT. 

**Additional context**

To fix, unconditionally set hi2c->XferSize =0u  in HAL_I2C_Mem_Write_IT.

**Screenshots**
If applicable, add screenshots to help explain your problem.

RJMSTM, 2023-05-11:
Hello @thealastair,

Thank you for this report. We will get back to you as soon as we analyze it further. This may take some time. Thank you for your comprehension.

 With regards,

HBOSTM, 2023-05-16:
Hello @thealastair ,

Thank you for this contribution, this point has been reported to our development teams. I will get back to you as soon as I have any updates.

Best Regards,

HBOSTM, 2023-05-16:
ST Internal Reference:  152242

closed 2023-07-21 by HBOSTM