STMicroelectronics/stm32l4xx-hal-driver #10: STM32L476RCT6 use freertos and sdio 4bits and dma,can not receive queue

Barfell, 2023-10-25:
STM32L476RCT6, use CUBEMX generate the project, use FREERTOS ,SDIO 4bits, DMA, and Fatfs, the SD not work normally, the library is STM32Cube_FW_L4_V1.18.0, the question repetition on other library 
in fuction
 DRESULT SD_read(BYTE lun, BYTE *buff, DWORD sector, UINT count), 
    event = osMessageGet(SDQueueID, SD_TIMEOUT);

can not receive message.

but use sdio 1bits，work normally.

Barfell, 2023-10-25:
set a break in the function void SDMMC1_IRQHandler(void),then full running,the sd init success.why?

TOUNSTM, 2024-01-17:
Hello @Barfell,

Thank you for your contribution. Regarding the point for which you opened the issue, as it is resolved, please allow me to close the issue.

With regards,

closed 2024-01-17 by TOUNSTM