STMicroelectronics/stm32l4xx-hal-driver #13: I2C: Mixed transfers methods causing Interrupts to hang MCU

flatirondesigns, 2024-01-02:
STM32CubeIDE v 1.14.0
NUCLEO-L412KB

This issue appears to be the same as
https://github.com/STMicroelectronics/STM32CubeH7/issues/260#top

If I do a DMA transfer over I2C then followed by HAL_I2C_Mem_Write, it hangs.  The TXIE is still set after the DMA transfer. If I manually clear it through the IDE before executing the Mem_Write, it does not hang.

RJMSTM, 2024-01-04:
Hello @flatirondesigns,

Thank you for this report, actually this issue has already been detected and fixed. The fix is already available in the frame of the v1.18.0 release published a couple of weeks ago. The linking error should disappear once you will upgrade to this version.

Thank you again for your contribution.
With regards,

flatirondesigns, 2024-01-04:
Oh, sorry. I didn't see a new version. I will check again. Thank you.

flatirondesigns, 2024-01-04:
Sorry, I didn't realize there was a separate menu option to download the latest FW. It does fix my issue. Thanks again.

closed 2024-02-05 by ALABSTM