STMicroelectronics/stm32l4xx-hal-driver #14: Incorrect STM32L4P5xx and STM32L4Q5xx RNG Initialisation in Low Level API

badcf00d, 2024-01-16:
In a post here https://community.st.com/t5/stm32-mcus-security/why-am-i-getting-an-rng-failure-on-the-stm32l4p5/m-p/203250. It was recognised that the HAL was initialising the RNG peripheral incorrectly.

The HAL function `HAL_RNG_Init` seems to have been corrected to set the correct CONFIG[1,2,3] and HTCR.

However the LL version of this function doesn't include any of this, so the CONFIG[1,2,3] and HTCR will all contain the wrong values when using this function.

https://github.com/STMicroelectronics/stm32l4xx_hal_driver/blob/eb4b2d2abad64c85000fb883a9fb86ab9fa07a91/Src/stm32l4xx_ll_rng.c#L118-L135

Could this function be updated to set the correct values, as described in Table 225. RNG configurations
of RM0432?

badcf00d, 2024-01-16:
Another suggestion, it would be useful if `LL_RNG_SetHealthconfiguration` was modified to write the magic bytes `0x17590ABC` to unlock the HTCR register before writing the input `HTCFG`

https://github.com/STMicroelectronics/stm32l4xx_hal_driver/blob/eb4b2d2abad64c85000fb883a9fb86ab9fa07a91/Inc/stm32l4xx_ll_rng.h#L661-L664

Currently this function will have no effect if the user just calls it with the value they want in the HTCR e.g.

```c
LL_RNG_SetHealthconfiguration(0x00007274);
```

RJMSTM, 2024-01-19:
Hello @badcf00d,

Thank you for your contribution. Your report will be forwarded to our development team for analysis.
We will get back to you as soon as we have an answer.

As a first answer, it's Cleary mentioned in Reference manual to unlock the HCR register before reading or writing in the register in this case you can call the function `LL_RNG_SetHealthconfiguration` two times firstly write the magic bytes **0x17590ABC** to unlock the HTCR register and then you can read or write this register.

Thanks in advance for your patience and understanding.

With Regards,

RJMSTM, 2024-02-09:
ST Internal Reference: 173093