STMicroelectronics/stm32l4xx-hal-driver #15: HAL_RTCEx_SetTamper_IT Incorrect on STM32L4P5xx and STM32L4Q5xx

badcf00d, 2024-02-27:
The HAL function `HAL_RTCEx_SetTamper_IT` does not initialise the trigger level properly in `TAMP->CR2`, the function `HAL_RTCEx_SetTamper` seems correct though. I suspect the updated code from `HAL_RTCEx_SetTamper` just needs to be copied across to `HAL_RTCEx_SetTamper_IT`.

Correct:
https://github.com/STMicroelectronics/stm32l4xx_hal_driver/blob/e49465048ae4329091e031dd9ae93fed1c2e3389/Src/stm32l4xx_hal_rtc_ex.c#L1767-L1779

Incorrect:
https://github.com/STMicroelectronics/stm32l4xx_hal_driver/blob/e49465048ae4329091e031dd9ae93fed1c2e3389/Src/stm32l4xx_hal_rtc_ex.c#L1928-L1936

ALABSTM, 2024-02-29:
Hi @badcf00d,

Thank you for having reported. Looks like you're right. The issue will be forwarded to our development teams to be fixed. I will keep you informed.

With regards,

ALABSTM, 2024-02-29:
ST Internal Reference: 174977

KRASTM, 2024-03-29:
Fixed in 67bb79eb5732c8a46ee51ae9db255a487f257600

closed 2024-03-29 by KRASTM