STMicroelectronics/stm32l4xx-hal-driver #18: HAL_SPI_Receive with incorrect length/data pointer parameters renders state stuck in HAL_SPI_STATE_BUSY_RX

mmarciniewski, 2024-06-26:
**Describe the set-up**
Any board, compiler should be logically irrelevant, but tested with GCC 13.2.1
Tested on commit e49465048ae4329091e031dd9ae93fed1c2e3389

**Describe the bug**
When using the HAL_SPI_Receive APIs including IT and DMA derivatives, with a Size/Length of 0, the driver renders the SPI handle stuck in HAL_SPI_STATE_BUSY_RX state after returning an error, preventing further use of the handle. This should also apply if passing NULL pData parameter. 

While a 0 input is incorrect, This should reasonably not block further use of the handle after returning an error.

The cause is that the state is set to HAL_SPI_STATE_BUSY_RX on lines 1023, 1689, 2022 in stm32l4xx_hal_spi.c, for the corresponding Polling/Interrupt/DMA Receive API calls, followed by a immediate return of HAL_SPI_TransmitReceive (and IT/DMA type calls).

The state is then never reset before returning an error in the TransmitReceive API on lines 1293, 1803, 2175.

**How To Reproduce**
Call HAL_SPI_Receive API with a length of 0 at least once, followed by any other API call dependant on the state being ready, such as another Transmit/Receive.

KRASTM, 2024-06-28:
ST Internal Reference: 185358

KRASTM, 2024-07-23:
Fixed in commit https://github.com/STMicroelectronics/stm32l4xx_hal_driver/commit/41027c61f69401afe4937db0ebd5d0cdbffbea0d

closed 2024-07-23 by KRASTM