STMicroelectronics/stm32l4xx-hal-driver #21: I2C_TransferConfig Mode Sets bit in SADD?

masonbutler, 2025-03-26:
I'm using an STM32L452RE-Nucleo and I noticed this going through the headers:

It appears that in the I2C modules in stm32l4xx_hal_i2c.c, I2C_TransferConfig accepts an argument called Mode which gets set in tmp into the slave address section of the register. This doesn't look right unless there is some behaviour I'm not aware of? I looked at the ref manual to see but there is no bit in reg I2C_CR2 that would be set for this. Also shouldn't this be clearing the top 5 reserve bits of the register when masking?

```
static void I2C_TransferConfig(I2C_HandleTypeDef *hi2c, uint16_t DevAddress, uint8_t Size, uint32_t Mode,
                               uint32_t Request)
{
  uint32_t tmp;

  /* Check the parameters */
  assert_param(IS_I2C_ALL_INSTANCE(hi2c->Instance));
  assert_param(IS_TRANSFER_MODE(Mode));
  assert_param(IS_TRANSFER_REQUEST(Request));

  /* Declaration of tmp to prevent undefined behavior of volatile usage */
  tmp = ((uint32_t)(((uint32_t)DevAddress & I2C_CR2_SADD) | \
                    (((uint32_t)Size << I2C_CR2_NBYTES_Pos) & I2C_CR2_NBYTES) | \
                    (uint32_t)Mode | (uint32_t)Request) & (~0x80000000U));

  /* update CR2 register */
  MODIFY_REG(hi2c->Instance->CR2, \
             ((I2C_CR2_SADD | I2C_CR2_NBYTES | I2C_CR2_RELOAD | I2C_CR2_AUTOEND | \
               (I2C_CR2_RD_WRN & (uint32_t)(Request >> (31U - I2C_CR2_RD_WRN_Pos))) | \
               I2C_CR2_START | I2C_CR2_STOP)), tmp);
}
```
The constant is defined here, so 0x10  would be set in the middle of the SADD register. In fact, none of these constants make sense given the register layout in the manual as they all set bits in the address.
```
typedef enum
{
  HAL_I2C_MODE_NONE               = 0x00U,   /*!< No I2C communication on going             */
  HAL_I2C_MODE_MASTER             = 0x10U,   /*!< I2C communication is in Master Mode       */
  HAL_I2C_MODE_SLAVE              = 0x20U,   /*!< I2C communication is in Slave Mode        */
  HAL_I2C_MODE_MEM                = 0x40U    /*!< I2C communication is in Memory Mode       */

} HAL_I2C_ModeTypeDef;
```

An example of it being called is in:
```
HAL_StatusTypeDef HAL_I2C_Master_Transmit(I2C_HandleTypeDef *hi2c, uint16_t DevAddress, uint8_t *pData,
                                          uint16_t Size, uint32_t Timeout)
{
  uint32_t tickstart;
  uint32_t xfermode;

  if (hi2c->State == HAL_I2C_STATE_READY)
  {
    /* Process Locked */
    __HAL_LOCK(hi2c);

    /* Init tickstart for timeout management*/
    tickstart = HAL_GetTick();

    if (I2C_WaitOnFlagUntilTimeout(hi2c, I2C_FLAG_BUSY, SET, I2C_TIMEOUT_BUSY, tickstart) != HAL_OK)
    {
      return HAL_ERROR;
    }

    hi2c->State     = HAL_I2C_STATE_BUSY_TX;
    hi2c->Mode      = HAL_I2C_MODE_MASTER;
    hi2c->ErrorCode = HAL_I2C_ERROR_NONE;

    /* Prepare transfer parameters */
    hi2c->pBuffPtr  = pData;
    hi2c->XferCount = Size;
    hi2c->XferISR   = NULL;

    if (hi2c->XferCount > MAX_NBYTE_SIZE)
    {
      hi2c->XferSize = MAX_NBYTE_SIZE;
      xfermode = I2C_RELOAD_MODE;
    }
    else
    {
      hi2c->XferSize = hi2c->XferCount;
      xfermode = I2C_AUTOEND_MODE;
    }

    if (hi2c->XferSize > 0U)
    {
      /* Preload TX register */
      /* Write data to TXDR */
      hi2c->Instance->TXDR = *hi2c->pBuffPtr;

      /* Increment Buffer pointer */
      hi2c->pBuffPtr++;

      hi2c->XferCount--;
      hi2c->XferSize--;

      /* Send Slave Address */
      /* Set NBYTES to write and reload if hi2c->XferCount > MAX_NBYTE_SIZE and generate RESTART */
      I2C_TransferConfig(hi2c, DevAddress, (uint8_t)(hi2c->XferSize + 1U), xfermode,
                         I2C_GENERATE_START_WRITE);
```

The register is defined in the ref manual as:

[image]

Either I'm not seeing something, something is undocumented, or this is a mistake. Could someone please clarify?

ALABSTM, 2025-04-02:
Hi @masonbutler,

It looks like there is a confusion. According to the description of parameter `Mode`, the possible values are those indicated below:

https://github.com/STMicroelectronics/stm32l4xx-hal-driver/blob/a60c3480b58ea4f9704fd1c72ce77a6358662c5c/Src/stm32l4xx_hal_i2c.c#L7349-L7353

These possible values are defined already shifted the right number of bits so they directly fall on bits **24** and **25** of the `CR2` register.

https://github.com/STMicroelectronics/stm32l4xx-hal-driver/blob/a60c3480b58ea4f9704fd1c72ce77a6358662c5c/Inc/stm32l4xx_hal_i2c.h#L389-L391

I hope this makes things clearer. Do not hesitate, should you still have questions.

With regards,

masonbutler, 2025-04-03:
You are correct, I was confusing device HAL mode:

`    hi2c->Mode      = HAL_I2C_MODE_MASTER;
`

With the actual transfer HW mode:

`xfermode = I2C_RELOAD_MODE;
`

Thank you

closed 2025-04-03 by masonbutler